luar123/esphome_zb_sensor #1: Use newer version of ESP-IDF as required IDF version for better ESP32-C6 and ESP32-H2 support?

Hedda, 2024-03-26:
Suggest set a higher minimum requirement for ESP-IDF version since 5.0.0 does not have "good" support for ESP32-C6/ESP32-H2?

ESP-IDF v5.0.0 was first version to support ESP32-C6 and ESP32-H2, but they do not have "good" support until ESP-IDF v5.1 or 5.2

luar123, 2024-03-26:
Not sure what exactly you are talking about. In ZB_sensor.yaml the esp-idf version is set to 5.1.2.

lboue, 2024-03-26:
> Not sure what exactly you are talking about. In ZB_sensor.yaml the esp-idf version is set to 5.1.2.

@luar123 
Here:
https://github.com/luar123/esphome_zb_sensor/blob/master/idf_component.yml#L8

Hedda, 2024-03-26:
> Here:
> https://github.com/luar123/esphome_zb_sensor/blob/master/idf_component.yml#L8

Yeah that is what I meant. There is currently require 5.0.0 or newer

lboue, 2024-03-26:
I make PR #7 for that.

commit eb560655 (2024-03-26): Bump esp-zigbee-lib and esp-zboss-lib to 1.2.1. Set esp-idf>=5.1.2
Fix #6
Fix #1
closed 2024-03-26 by luar123