luar123/esphome_zb_sensor #3: `src/esp_zb_light.c:43:10: fatal error: ha/esp_zigbee_ha_standard.h: No such file or directory`

lboue, 2024-03-26:
I am facing an issue compiling ZB_sensor.yaml file. It doesn't find the library file: `ha/esp_zigbee_ha_standard.h`
`src/esp_zb_light.c:43:10: fatal error: ha/esp_zigbee_ha_standard.h: No such file or directory`

```
(venv) user@laptop:~/esphome$ esphome run ZB_sensor.yaml 
INFO ESPHome 2024.4.0-dev
INFO Reading configuration ZB_sensor.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing zb-sensor (board: esp32-h2-devkitm-1; framework: espidf; platform: https://github.com/luar123/platform-espressif32H2/archive/refs/tags/v6.5.1.zip)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32H2 96MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50102.240122 (5.1.2) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208
Reading CMake configuration...
No dependencies
Compiling .pioenvs/zb-sensor/src/esp_zb_light.o
src/esp_zb_light.c:43:10: fatal error: ha/esp_zigbee_ha_standard.h: No such file or directory
   43 | #include "ha/esp_zigbee_ha_standard.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pioenvs/zb-sensor/src/esphome/components/logger/logger_rp2040.o
Compiling .pioenvs/zb-sensor/src/esphome/components/sensor/automation.o
*** [.pioenvs/zb-sensor/src/esp_zb_light.o] Error 1
Compiling .pioenvs/zb-sensor/src/esphome/components/sensor/filter.o
============================================================================================== [FAILED] Took 8.57 seconds ==============================================================================================
```

luar123, 2024-03-26:
I will look into this when I am at my computer. But there shouldn't be any arduino ZigBee libs as everything is ESP-IDF based.

lboue, 2024-03-26:
> I will look into this when I am at my computer. But there shouldn't be any arduino ZigBee libs as everything is ESP-IDF based.

It look like idf_component.yml is not taken into account because we have this log: `No dependencies`
```
INFO ESPHome 2024.4.0-dev
INFO Reading configuration ZB_sensor.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing zb-sensor (board: esp32-h2-devkitm-1; framework: espidf; platform: https://github.com/luar123/platform-espressif32H2/archive/refs/tags/v6.5.1.zip)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32H2 96MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50102.240122 (5.1.2) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208
Reading CMake configuration...
No dependencies
```

luar123, 2024-03-26:
Ok, please check if there is a `managed_components`folder in `.esphome/build/zb-sensor`containing the zigbee libs.

If not you could try to use docker, just to make sure there is nothing wrong with you local setup:

`docker run --rm -v "${PWD}":/config -it ghcr.io/esphome/esphome:dev run ZB_sensor.yaml `

First run:

```
src/esp_zb_light.c:43:10: fatal error: ha/esp_zigbee_ha_standard.h: No such file or directory
   43 | #include "ha/esp_zigbee_ha_standard.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
Copy `idf_component.yml` and sdk configs.
Second run should then succeed.

luar123, 2024-03-26:
And start from a clean environment (deleting at least .esphome/build/zb-sensor/.pioenvs or better the whole .esphome folder)

lboue, 2024-03-26:
Thanks. I will try this workaround and let you know.

lboue, 2024-03-26:
Thanks. I had copy idf_component.yml file in the wrong directory for the first try.
It works

```

Generating partitions .pioenvs/zb-sensor/partitions.bin
Generating an empty partition .pioenvs/zb-sensor/ota_data_initial.bin
Archiving .pioenvs/zb-sensor/bootloader/esp-idf/spi_flash/libspi_flash.a
Indexing .pioenvs/zb-sensor/bootloader/esp-idf/spi_flash/libspi_flash.a
Archiving .pioenvs/zb-sensor/bootloader/esp-idf/micro-ecc/libmicro-ecc.a
Indexing .pioenvs/zb-sensor/bootloader/esp-idf/micro-ecc/libmicro-ecc.a
Linking .pioenvs/zb-sensor/bootloader.elf
Building .pioenvs/zb-sensor/bootloader.bin
Creating esp32h2 image...
Successfully created esp32h2 image.
Linking .pioenvs/zb-sensor/firmware.elf
RAM:   [=         ]   9.6% (used 31428 bytes from 327680 bytes)
Flash: [===       ]  30.8% (used 545778 bytes from 1769472 bytes)
Building .pioenvs/zb-sensor/firmware.bin
Creating esp32h2 image...
Successfully created esp32h2 image.
esp32_create_combined_bin([".pioenvs/zb-sensor/firmware.bin"], [".pioenvs/zb-sensor/firmware.elf"])
Wrote 0x97bc0 bytes to file /home/ludovic/esphome/.esphome/build/zb-sensor/.pioenvs/zb-sensor/firmware-factory.bin, ready to flash to offset 0x0
============================================================================================ [SUCCESS] Took 211.59 seconds ============================================================================================
INFO Successfully compiled program.

```

closed 2024-03-26 by lboue