luar123/esphome_zb_sensor #6: Bump esp-zigbee-lib 1.2.1 and esp-zboss-lib to 1.2.1 or not?

Hedda, 2024-03-26:
Latest versions is currently esp-zigbee-lib 1.2.1 and esp-zboss-lib to 1.2.1 so question wheater or not to update?

https://components.espressif.com/components/espressif/esp-zigbee-lib

https://components.espressif.com/components/espressif/esp-zboss-lib

Unclear what exactly is fixed that is specifically related to this project but upstream looks to currently be very active with fixes + more:

https://github.com/espressif/esp-zigbee-sdk/commit/ba8582df6bc62b9e5e69a4c6f3ae02f0e1da194a

esp-zigbee-sdk: release/v1.2.1(aaf0078d)
- Updated esp-zboss-lib to v1.2.1
- Updated esp-zigbee-lib to v1.2.1
- Added manufacturer code setting API
- Supported the large data transmission for zcl commands
- Fixed mfg tool
- Fixed multiple link key initialization
- Fixed touchlink nwk update request handler
- Closes: https://github.com/espressif/esp-zigbee-sdk/issues/202
- Closes: https://github.com/espressif/esp-zigbee-sdk/issues/244
- Closes: https://github.com/espressif/esp-zigbee-sdk/issues/254

https://github.com/espressif/esp-zboss-lib/commit/4e46311f060a1a45e02caea4caead311f49e584c

esp-zboss-lib: release/v1.2.1(aaf0078d)
- Updated esp-zboss-lib to v1.2.1
- Supported large data transmission for zcl commands
- Fixed multi-link key initialization
- Fixed touchlink nwk update request handler

luar123, 2024-03-26:
Not sure why the idf example still used the old versions. Anyway, thanks for pointing that out. I will push an update as it is a rather easy fix.

closed 2024-03-26 by luar123

Hedda, 2024-03-28:
> Not sure why the idf example still used the old versions.

@luar123 note that it can be confusing to use "corect" latest upstream from Espressif when they have copies of examples in different places but not keeping them updated in all locations, so one question is if you are using the upstream examples from the esp-zigbee-sdk repository or from the esp-idf repository?

https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/

verses

https://github.com/espressif/esp-idf/tree/master/examples/zigbee

I believe that the former ESP Zigbee SDK should generally provide the most up to date examples for Zigbee

commit eb560655 (2024-03-26): Bump esp-zigbee-lib and esp-zboss-lib to 1.2.1. Set esp-idf>=5.1.2
Fix #6
Fix #1