luar123/esphome_zb_sensor #10: [SUGGESTION] Copy readme instructions from latest examples in ESP-Zigbee-SDK repository?

Hedda, 2024-03-28:
Suggest maybe consider copy some additional instruction steps from these very latest Zigbee examples in ESP-Zigbee-SDK repository?

* https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample
  * https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_temperature_sensor
  * https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_thermostat
  * https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_on_off_light
  * https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_color_dimmable_light

Note that readme for the respective Zigbee examples that are in the esp-idf repository are not all the same as in above esp-zigbee-sdk:

* https://github.com/espressif/esp-idf/tree/master/examples/zigbee/light_sample
  * https://github.com/espressif/esp-idf/tree/master/examples/zigbee/light_sample/HA_on_off_light
  * https://github.com/espressif/esp-idf/tree/master/examples/zigbee/light_sample/HA_on_off_switch

lboue, 2024-03-28:
I think we could the flashing step (with esptool?)

Hedda, 2024-03-28:
_Originally posted by @lboue in https://github.com/luar123/esphome_zb_sensor/issues/9#issuecomment-2024695517_
>   **_@Hedda I think we could the missing flashing step (with esptool?)_**

Do you mean also add another section to readme called samething like "How to flash using esptool"

https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flashing-firmware.html

https://github.com/espressif/esptool

...and not flashing with ESP Web Tools?

https://github.com/esphome/esp-web-tools

https://web.esphome.io/

lboue, 2024-03-28:
> _Originally posted by @lboue in [#9 (comment)](https://github.com/luar123/esphome_zb_sensor/pull/9#issuecomment-2024695517)_
> 
> > **_@Hedda I think we could the missing flashing step (with esptool?)_**
> 
> Do you mean also add another section to readme called samething like "How to flash using esptool"
> 
> https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flashing-firmware.html
> 
> https://github.com/espressif/esptool
> 
> ...and not flashing with "ESP Web Tools" which I believe is the general recommedation for ESPHome?
> 
> https://github.com/esphome/esp-web-tools
> 
> https://esphome.github.io/esp-web-tools/
> 
> https://web.esphome.io/
> 
> Update: Perhaps esp-web-tools does not yet have support or ESP32-H2 and/or ESP32-C6? There is an open issue about those:
> 
> [esphome/esp-web-tools#455](https://github.com/esphome/esp-web-tools/issues/455)
> 
> Did not check as I have only done research and paper exercise so far, but I do not see why esp-web-tools would not work?

I haven't yet received my ESP32-H2 board, so I haven't been able to check compatibility. 
I think there should be an step instruction about "how to flash". Either with the _esp-web-tools_ or in CLI with _esptool_ .

luar123, 2024-03-28:
I would not add further information like flashing or sdkconfig to this project as this should all be handled by esphome. 
I think the links to the examples you have added already are enough.

closed 2024-04-04 by luar123