luar123/esphome_zb_sensor #12: Suggest update example to ESP-IDF 5.2.1 + esp-zboss-lib + esp-zigbee-lib "~1.2.1" which currently are very latest versions.

Hedda, 2024-03-28:
As per https://github.com/luar123/esphome_zb_sensor/issues/6 but not quite implemented in https://github.com/luar123/esphome_zb_sensor/commit/eb560655323ae8df8ae987efc279aed4fa413e6e suggest update IDF Component Manager Manifest File to the latest stable releases:

https://github.com/luar123/esphome_zb_sensor/blob/master/idf_component.yml

```
## IDF Component Manager Manifest File
dependencies:
  espressif/esp-zigbee-lib: "~1.2.1"
  espressif/esp-zboss-lib: "~1.2.1"
  ## Required IDF version
  idf:
    version: ">=5.2.1"
```

Suggest update this example to ESP-IDF release v5.2.1 + esp-zboss-lib + esp-zigbee-lib "~1.2.1" which currently are very latest versions.

ESP-IDF Release v5.2.1 is currently the latest stable IDF release:

https://github.com/espressif/esp-idf/releases/tag/v5.2.1

esp-zigbee-lib 1.2.1 is currently the latest stable esp-zigbee-lib release:

https://components.espressif.com/components/espressif/esp-zigbee-lib

esp-zboss-lib 1.2.1 is currently the latest stable esp-zboss-lib release:

https://components.espressif.com/components/espressif/esp-zboss-lib

luar123, 2024-03-28:
Once esphome and platformio is compatible with IDF 5.1.3 or 5.2 I will update.
Feel free to try if you can make ZigBee-lib 1.2.1 work with IDF 5.1.2.

lboue, 2024-03-28:
@luar123 
It works with my ESP32-C6 board:

[image]

[image]

luar123, 2024-04-04:
Testing esp-idf 5.2.1 right now with pio 6.6.0. Seems to work in general, but there is one issue in esphome that needs to get solved first.

lboue, 2024-04-04:
> Testing esp-idf 5.2.1 right now with pio 6.6.0. Seems to work in general, but there is one issue in esphome that needs to get solved first.

Do you have the issue number?

jjcszxh, 2024-04-09:
> IDF 5.1.2 适用于我的 [ESP32-C6-DevKitC-1-N8](https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/index.html) 开发板：
> 
> ![图像](https://private-user-images.githubusercontent.com/938089/317830242-9fb76eed-3013-4a13-8060-bfdf0bf1c8f5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI2NjYxNzcsIm5iZiI6MTcxMjY2NTg3NywicGF0aCI6Ii85MzgwODkvMzE3ODMwMjQyLTlmYjc2ZWVkLTMwMTMtNGExMy04MDYwLWJmZGYwYmYxYzhmNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQwOVQxMjMxMTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jNTQyMzZkZTVjY2E5ZThlMTM5NDNmNTM2MWIyNzg5M2JkMTllMTA3NTQ0MThhZWE0ZmFhYjJlODRhMTk2OGNjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ovnhhDpgXB8ph-mgGuuhCGD9JQ9saFljeG3JNYvRi5w)
> 
> ![图像](https://private-user-images.githubusercontent.com/938089/317830453-97a7f27e-f0ba-4eed-8594-db3b3309c97a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI2NjYxNzcsIm5iZiI6MTcxMjY2NTg3NywicGF0aCI6Ii85MzgwODkvMzE3ODMwNDUzLTk3YTdmMjdlLWYwYmEtNGVlZC04NTk0LWRiM2IzMzA5Yzk3YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQwOVQxMjMxMTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mZjQ4NTk5ODRmYjc3ODI3MDcwMTU2MWEwZWM2YTU1ZmEzOTg4OTIyN2U5ZmQ4NDAzODRhMWY1N2YxZGY4MWI1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.0vKZnZwFhqboIRe7hoBUU_gGb_Dv8vo8gZ9KBDm1noI)

How did you get ESP32c6 to support Zigbee? Thank you

Hedda, 2024-04-09:
> How did you get ESP32c6 to support Zigbee? Thank you

What do you mean and is that question on-topic for the
this specific issue in the original post above?

Please see README.md -> https://github.com/luar123/esphome_zb_sensor/blob/master/README.md

For more references also see this feature request -> https://github.com/esphome/feature-requests/issues/1397

In summery; this depends on Espressif Zigbee SDK -> https://github.com/espressif/esp-zigbee-sdk

closed 2024-06-08 by luar123