luar123/esphome_zb_sensor #14: AHT10 bug fix and changes in ESPHome 2024.3.1 (released on 27th of March 2024)

Hedda, 2024-03-28:
FYI, there was a bug fix and a couple of changes to AHT10 inside ESPHome 2024.3.1 (released on 27th of March 2024)

https://esphome.io/changelog/2024.3.0.html

https://github.com/esphome/esphome/pull/6409

https://github.com/esphome/esphome/pull/6401

https://github.com/esphome/esphome/pull/6303

luar123, 2024-03-28:
Logger fix is not included in 2024.03.1

lboue, 2024-03-29:
@Hedda 
What do you get in debug mode?

I just tried ESPHome firmware 2024.3.1 with ESP32-S3-WROOM-1-N16R8 with a [AHT20](https://esphome.io/components/sensor/aht10.html)  sensor and it works:

[image]

lboue, 2024-04-05:
Setting up AHT20 with ESP32H works too
```
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyACM0 with baud rate 115200
[21:38:51]\xf8\xf8\xf8\xf8x\x80\xf8\xf8\x80\xf8\x80\x80\x80\x80\x80x\xf8x\x80\x80x\x80x\x80\x80\x80\xf8\x80\x80xx\x80\x80x\xf8x\xf8x\xf8x\xf8\x80\x80xx\x80xxxxx\x80xx\xf8[0m
[21:38:51]I (24) boot: compile time Apr  5 2024 21:37:20
[21:38:51]I (25) boot: chip revision: v0.1
[21:38:51]I (26) boot.esp32h2: SPI Speed      : 64MHz
[21:38:51]I (31) boot.esp32h2: SPI Mode       : DIO
[21:38:51]I (36) boot.esp32h2: SPI Flash Size : 4MB
[21:38:51]I (41) boot: Enabling RNG early entropy source...
[21:38:51]I (46) boot: Partition Table:
[21:38:51]I (50) boot: ## Label            Usage          Type ST Offset   Length
[21:38:51]I (57) boot:  0 otadata          OTA data         01 00 00009000 00002000
[21:38:51]I (64) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[21:38:51]I (72) boot:  2 app0             OTA app          00 10 00010000 001b0000
[21:38:51]I (79) boot:  3 app1             OTA app          00 11 001c0000 001b0000
[21:38:51]I (87) boot:  4 nvs              WiFi data        01 02 00370000 0006d000
[21:38:51]I (94) boot:  5 zb_storage       Unknown data     01 81 003dd000 00004000
[21:38:51]I (102) boot:  6 zb_fct           Unknown data     01 81 003e1000 00000400
[21:38:51]I (110) boot: End of partition table
[21:38:51]I (114) boot: No factory image, trying OTA 0
[21:38:51]I (119) esp_image: segment 0: paddr=00010020 vaddr=42070020 size=11cf8h ( 72952) map
[21:38:51]I (150) esp_image: segment 1: paddr=00021d20 vaddr=40800000 size=0c3f4h ( 50164) load
[21:38:51]I (167) esp_image: segment 2: paddr=0002e11c vaddr=00000000 size=01efch (  7932) 
[21:38:51]I (171) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=6c848h (444488) map
[21:38:51]I (311) boot: Loaded app from partition at offset 0x10000
[21:38:51]I (332) boot: Set actual ota_seq=1 in otadata[0]
[21:38:51]I (333) boot: Disabling RNG early entropy source...
[21:38:51]I (343) cpu_start: Unicore app
[21:38:51]I (344) cpu_start: Pro cpu up.
[21:38:51]W (353) clk: esp_perip_clk_init() has not been implemented yet
[21:38:51]I (359) cpu_start: Pro cpu start user code
[21:38:51]I (360) cpu_start: cpu freq: 96000000 Hz
[21:38:51]I (360) cpu_start: Application information:
[21:38:51]I (362) cpu_start: Project name:     zb-sensor
[21:38:51]I (367) cpu_start: App version:      2024.4.0-dev
[21:38:51]I (373) cpu_start: Compile time:     Apr  5 2024 21:35:21
[21:38:51]I (379) cpu_start: ELF file SHA256:  5aff87ecbf58e88f...
[21:38:51]I (385) cpu_start: ESP-IDF:          5.1.2
[21:38:51]I (390) cpu_start: Min chip rev:     v0.0
[21:38:51]I (394) cpu_start: Max chip rev:     v0.99 
[21:38:51]I (399) cpu_start: Chip rev:         v0.1
[21:38:51]I (404) heap_init: Initializing. RAM available for dynamic allocation:
[21:38:51]I (411) heap_init: At 40815290 len 000380F0 (224 KiB): D/IRAM
[21:38:51]I (417) heap_init: At 4084D380 len 00002B60 (10 KiB): STACK/DIRAM
[21:38:51]I (425) spi_flash: detected chip: generic
[21:38:51]I (429) spi_flash: flash io: dio
[21:38:51]I (433) sleep: Configure to isolate all GPIO pins in sleep state
[21:38:51]I (440) sleep: Enable automatic switching of GPIO sleep configuration
[21:38:51]I (447) app_start: Starting scheduler on CPU0
[21:38:51]I (452) main_task: Started on CPU0
[21:38:51]I (456) main_task: Calling app_main()
[21:38:52]I (611) main_task: Returned from app_main()
[21:38:52][I][logger:158]: Log initialized
[21:38:52][W][app:022]: Component custom_component already registered! (0x4081be74)
[21:38:52][I][app:029]: Running through setup()...
[21:38:52][C][i2c.idf:017]: Setting up I2C bus...
[21:38:52][I][i2c.idf:233]: Performing I2C bus recovery
[21:38:52][D][esp-idf:000]: I (619) gpio: GPIO[4]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
[21:38:52]
[21:38:52][D][esp-idf:000]: I (629) gpio: GPIO[5]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
[21:38:52]
[21:38:52][C][aht10:046]: Setting up AHT20
[21:38:52][D][esp-idf:000]: I (719) phy: phy_version: 211,0, 5857fe5, Nov  1 2023, 11:31:09
[21:38:52]
[21:38:52][D][esp-idf:000]: I (720) phy: libbtbb version: ce629d6, Nov  1 2023, 11:31:19
[21:38:52]
[21:38:52][I][app:062]: setup() finished successfully!
[21:38:52][D][esp-idf:000]: I (732) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[... 8 lines cut ...]
[21:38:52][C][logger:169]:   Log Baud Rate: 115200
[21:38:52][C][logger:170]:   Hardware UART: UART0
[21:38:52][C][i2c.idf:061]: I2C Bus:
[21:38:52][C][i2c.idf:062]:   SDA Pin: GPIO5
[21:38:52][C][i2c.idf:063]:   SCL Pin: GPIO4
[21:38:52][C][i2c.idf:064]:   Frequency: 50000 Hz
[21:38:52][C][i2c.idf:067]:   Recovery: bus successfully recovered
[21:38:52][I][i2c.idf:077]: Results from i2c bus scan:
[21:38:52][I][i2c.idf:083]: Found i2c device at address 0x38
[21:38:52][I][i2c.idf:083]: Found i2c device at address 0x77
[21:38:52][D][sensor:093]: 'Living Room Temperature': Sending state 23.50788 °C with 2 decimals of accuracy
[21:38:52][D][sensor:093]: 'Living Room Humidity': Sending state 60.25801 % with 2 decimals of accuracy
[21:38:52][C][aht10:160]: AHT10:
[21:38:52][C][aht10:161]:   Address: 0x38
[21:38:52][C][aht10:165]:   Temperature 'Living Room Temperature'
[21:38:52][C][aht10:165]:     Device Class: 'temperature'
[21:38:52][C][aht10:165]:     State Class: 'measurement'
[21:38:52][C][aht10:165]:     Unit of Measurement: '°C'
[21:38:52][C][aht10:165]:     Accuracy Decimals: 2
[21:38:52][C][aht10:166]:   Humidity 'Living Room Humidity'
[21:38:52][C][aht10:166]:     Device Class: 'humidity'
[21:38:52][C][aht10:166]:     State Class: 'measurement'
[21:38:52][C][aht10:166]:     Unit of Measurement: '%'
[21:38:52][C][aht10:166]:     Accuracy Decimals: 2
```

lboue, 2024-04-06:
> Logger fix is not included in 2024.03.1

[esphome 2024.3.2](https://github.com/esphome/esphome/pull/6472) is coming

lboue, 2024-04-10:
ESPHome (beta): 2024.4.0b1 is out with a fix:

- AHT10: fix temperature-only operation; add warning/error messages [esphome#6405](https://github.com/esphome/esphome/pull/6405) by [@clydebarrow](https://github.com/clydebarrow)