luar123/esphome_zb_sensor #24: Attempt to add illuminance cluster and sensor

dronix27, 2024-04-16:
Check this branch: https://github.com/dronix27/esphome_zb_sensor

lboue, 2024-04-16:
@dronix27

Shouldn't another file be created for this type of sensor to replace esp_zb_light.c?

dronix27, 2024-04-16:
> @dronix27
> 
> Shouldn't another file be created for this type of sensor to replace esp_zb_light.c?

Actually I don't know how to do it right.  I just added a new cluster to the same file.  I would be grateful if you tell me how to do this better.

luar123, 2024-04-16:
For now each ZigBee sensor is very individual because there is no posibility to configure through the yaml (I am working on it). There is no point in trying to generalize at this stage. So I would say your setup is fine. Of course you could rename the c file, but so could I 😃

Hedda, 2024-04-17:
> For now each ZigBee sensor is very individual because there is no posibility to configure through the yaml (I am working on it).

@luar123 tip there is to check out the ”Zigbee Home” project to see if maybe could port or reuse their config -> https://github.com/ffenix113/zigbee_home

closed 2024-06-08 by luar123