luar123/esphome_zb_sensor #28: New level for Zigbee in ESPHome

Muk911, 2024-05-10:
Hello, luar123.

I have prepared a small example with my Zigbee classes. You might be interested.

Best regards.

luar123, 2024-05-10:
Wow great, that looks like a lot of work. Some parts are very similar to my not yet published external ZigBee component (I need to fix a few issues before I want to publish).
I need to take a closer look on all your classes, but I could definitely reuse some parts. For example I didn't create classes for ep/clusters/etc but might need to in the future.
Do you plan to create a external component for esphome or keep it like this?

Muk911, 2024-05-10:
My main goal is C++ classes for Arduino/IDF. ESPHome components are just wrappers over the main classes and a way to use the existing ESPHome components for Zigbee devices.

I plan to figure out how to write external ESPHome components to simplify the configuration to this:
```
zigbee_device:
  platform: esp32_zigbee
  device_type: ESP_ZB_DEVICE_TYPE_ED
  
zigbee_endpoint:
  platform: esp32_zigbee
  ep_num: 1
  profile_id: ESP_ZB_AF_HA_PROFILE_ID
  device_id: ESP_ZB_HA_TEST_DEVICE_ID

zigbee_cluster:
  - platform: esp32_zigbee
    cluster_id: ESP_ZB_ZCL_CLUSTER_ID_BASIC
    role: ESP_ZB_ZCL_CLUSTER_SERVER_ROLE
    ep_num: 1
    attributes:
      - attr_id: ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID
        value: "MEA"
      - attr_id: ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID
        value: "myswitch"
      - attr_id: ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID
        value: 3
      - attr_id: ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID
        value: 4
      - attr_id: ESP_ZB_ZCL_ATTR_BASIC_APPLICATION_VERSION_ID
        value: 2

zigbee_cluster:
  - platform: esp32_zigbee
    cluster_id: ESP_ZB_ZCL_CLUSTER_ID_ON_OFF
    role: ESP_ZB_ZCL_CLUSTER_SERVER_ROLE
    ep_num: 1
    attributes:
      - attr_id: ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID
        id: za_on_off
        value: 0
        on_value:
          then:
            - lambda: |-
                if(x) {
                  id(light_1).turn_on();
                } else {
                  id(light_1).turn_off();
                }
      - attr_id: ESP_ZB_ZCL_ATTR_ON_OFF_ON_TIME
        id: za_on_time
        value: 3600
        on_value:
          then:
            - lambda: |-
                id(duration) = x;
      - attr_id: 0x3004
        type: ESP_ZB_ZCL_ATTR_TYPE_U16
        access: ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY
        value: 1200
```

luar123, 2024-05-13:
Sounds good!
I hope I will be able to publish my first version of the external component in the next weeks, maybe this could help you. It will be a little bit simpler, but still similar.

Hedda, 2024-05-16:
FYI, "zyaml" project for @hemtjanst  also look to contain a partial implementation of Zigbee specification in YAML representation:

https://github.com/hemtjanst/zyaml

Stumbled upon that project and not sure if it actually contains any concepts or ideas that could be usefor and borrowed from it?

Hedda, 2024-05-16:
Might you also be able to use the Connectivity Standards Alliance's project-chip (Matter) project's "ZAP" to generate clusters and attributes or other entities?

https://github.com/project-chip/zap

ZAP (which stands for "ZCL Advanced Platform" ) is a generic generation engine and user interface for applications and libraries based on Zigbee Cluster Library, the specification developed by the [Connectivity Standards Alliance](https://csa-iot.org/) to configure Matter and Zigbee Pro applications.

lxz946786639, 2024-05-27:
> 听上去很好！我希望我能够在接下来的几周内发布我的第一个版本的外部组件，也许这对您有所帮助。它会简单一点，但仍然相似。
> 
> 到目前为止，这是有效的：
> 
> ```
> sensor:
>   - platform: aht10
>     variant: AHT10
>     temperature:
>       name: "Living Room Temperature"
>       id: "temp"
>       on_value:
>         then:
>           - zigbee.setAttr:
>               id: zb
>               endpoint: 2
>               cluster: TEMP_MEASUREMENT
>               attribute: 0x0
>               value: !lambda "return x*100;"
>     update_interval: 60s
> 
> zigbee:
>   id: "zb"
>   endpoints:
>     - num: 1
>       device_type: COLOR_DIMMABLE_LIGHT
>       clusters:
>         - id: ON_OFF
>           attributes:
>             - id: 0
>               type: bool
>               on_value:
>                 then:
>                   - light.control:
>                       id: light_1
>                       state: !lambda "return (bool)x;"
>     - device_type: TEMPERATURE_SENSOR
>       num: 2
>       clusters:
>         - id: REL_HUMIDITY_MEASUREMENT
>           attributes:
>             - id: 0
>               type: U16
>               value: 20
>         - id: TEMP_MEASUREMENT
>           attributes:
>             - id: 0x0
>               type: S16
>               value: 10
>   on_join:
>     then:
>       - logger.log: "Joined network"
> 
> binary_sensor:
>   - platform: gpio
>     pin:
>       number: 9
>       mode:
>         input: true
>         pullup: true
>       inverted: true
>     id: button_1
>     on_click:
>       min_length: 5s
>       max_length: 20s
>       then:
>         - zigbee.reset: zb
> ```

Looking forward to your new version very much

luar123, 2024-06-08:
@lxz946786639: https://github.com/luar123/zigbee_esphome
@Muk911, I didn't re-use any of your work as I had most already working. But I could see a merge of both projects if you would like to go that way. I won't have time in the near future to look into this, though.

Hedda, 2024-06-08:
@luar123 @Muk911 @P-R-O-C-H-Y also check out Zigbee PoC for ESPHome on nRF52/nRF53 by @tomaszduda23 in https://github.com/tomaszduda23/esphome/pull/2